usdot-fhwa-stol/carma-platform #1650: When CLC follows ILC in a trajectory plan, the first CLC TrajectoryPlanPoint has an abnormally high speed

JonSmet, 2022-03-04:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Currently, plan_delegator shifts the starting and ending downtracks of each manuever in a received maneuver plan earlier by the distance between base_link and vehicle_front (Example: If the distance between base_link and vehicle_front is 4 meters, then an ILC maneuver received by plan_delegator will have its starting downtrack and ending downtrack decreased by 4 meters).

When a CLC maneuver follows an ILC maneuver in a maneuver plan, plan_delegator sends a trajectory plan request to the cooperative_lanechange plugin for the CLC maneuver using its updated starting and ending downtracks; however, the cooperative_lanechange plugin still uses the first point of the lane change starting lanelet, rather than the decreased maneuver starting downtrack provided by plan_delegator. 

Per trajectory planning design, the starting point of the CLC trajectory plan matches the same target_time as the last point of the previous ILC trajectory plan, and plan_delegator removes this "duplicate" point after the CLC trajectory plan is generated. However, now the trajectory plan includes a significant increase in downtrack distance from the last point of the ILC trajectory plan to the first point of the CLC trajectory plan, with a small increase in TrajectoryPlanPoint target_time that does not account for this large increase in distance. (Using the example from the first paragraph, this "jump" in distance would be ~4 meters).

### Carma version where this issue was discovered

CARMA Develop (as of February 28th, 2022)

### Expected behavior
If the speed limit for the ILC portion of the trajectory plan and the CLC portion of the trajectory plan are the same, the speed between the last ILC point and the first CLC point in the trajectory plan shall maintain this same speed, if road geometry allows. 

### Actual behavior

The speed between the final ILC trajectory plan point and the first CLC trajectory plan point is abnormally large.

### Steps to reproduce the actual behavior

1. Engage the CARMA System on a route that includes ILC maneuvers followed by a CLC maneuver.
2. Inspect the rosbag data to determine the speeds between sequential trajectory plan points by analyzing their (x,y) position and target_time.

### Related work

adev4a, 2022-03-04:
Based on the updated code CLC should be using the starting downtrack for the planning and not lanelet starting point https://github.com/usdot-fhwa-stol/carma-platform/blob/e1b105a64e64415b33be0e3a1d19a8b28c3252aa/basic_autonomy/src/basic_autonomy.cpp#L370
However the starting downtrack is being overwritten at the start of planning to the current downtrack of the vehicle. In case the plugin is requested to start planning early this might be causing an issue

JonSmet, 2022-03-29:
Partially fixed by #1671, fully fixed by #1685

closed 2022-03-29 by JonSmet