usdot-fhwa-stol/carma-platform #1666: Cannot Enable or Disable Plugins from the Web UI

GoodarzMehr, 2022-03-12:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

When running CARMA, guidance plugins cannot be enabled or disabled from the web UI. Nothing happens when clicking on a plugin's button. We want to enable the Platooning Strategic plugin in particular but we cannot currently do that. Our browser is Google Chrome.

### Carma version where this issue was discovered

develop

### Expected behavior

Guidance plugins become enabled or disabled from the web UI.

### Actual behavior

Nothing happens when clicking on a plugin's button.

### Steps to reproduce the actual behavior

1. Start CARMA.
2. Navigate to localhost on a web browser, then click START Platform.
3. Click on Advanced, then Change Plugins. Try enabling or disabling any of the plugins.

### Related work

msmcconnell, 2022-06-09:
Following up on this. It seems this functionality was never fully implemented in carma3. It appears to have been an after thought since it was functional in carma2.  
The functionality is currently being added back for ROS2 plugins and will probably make it into release 4.2 or worst case 4.3.
There are not currently plans to implement this for ROS1 plugins or carma3, but users can manually disable plugins by modifying the carma/launch/guidance.launch file.

GoodarzMehr, 2022-06-10:
Thanks for the clarification.