usdot-fhwa-stol/carma-platform #1707: CARMA Platform fails to generate trajectory near end of maneuver distances

kjrush, 2022-04-06:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

The trajectory generation logic in basic_autonomy fails to compute a trajectory when the vehicle's current pose (at time of planning) is close to the end of the trajectory (within curve-resampling distance of the last two points of the associated maneuver).  This causes a check within basic_autonomy to fail and return an empty trajectory.

### Carma version where this issue was discovered

CARMA Platform 3.8.0
### Expected behavior
The basic_autonomy logic should return at minimum the current pose and the end pose of the trajectory as valid trajectory points for any trajectory plan where the vehicle has not yet crossed the end of the maneuver.

A more proper fix later would involve implementing more sophisticated handling of interaction between the plan delegator and the tactical plugins to indicate maneuver execution completion.

### Actual behavior

The ILC plugin fails to generate trajectory causing a system exception.

### Steps to reproduce the actual behavior

The issue is largely stochastic in it's appearance, as it depends on the timing of the planning window with respect to the vehicle's pose at the end of the trajectory. But it does appear repeatedly in XIL simulation.

### Related work

This issue should be fixed in #1696

msmcconnell, 2022-04-26:
Resolved by linked PR in "Related Work"

msmcconnell, 2022-04-26:
Reopening to represent the value in implementing the more sophisticated fix. But reducing severity to 3