usdot-fhwa-stol/carma-platform #1725: Improving the entering time (ET) estimation in lci_strategic_plugin

Saeid-Solei, 2022-04-14:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Currently, the vehicle adds a configurable time buffer to the estimated entering time (ET) for the first time after it enters the Trajectory Smoothing (TS) area regardless of whether the initially estimated ET is within red, yellow, or green phase. To guarantee that the vehicle will not enter the intersection box at the end of a red phase and before the signal light changes from red to green, this time buffer must be added if the initially estimated ET is within a yellow\red phase (i.e., if the vehicle must enter the intersection box at the beginning of a green phase). However, if the initially estimated ET is within a green phase, there is no need to add the time buffer to the estimated ET. This causes the vehicle to slow down while it can continue moving with the speed limit and enter the intersection box in a middle of a green phase. To fix this issue, a check must be added to the code to add the time buffer to the initially estimated ET only if it is within a yellow\red phase.

### Carma version where this issue was discovered

CARMA 4.0.2

### Expected behavior

If the vehicle can continue with the speed limit and enter the intersection box during a green phase (i.e., if the initially estimated ET is already within a green phase), the time buffer shall not be added to the initially estimated ET and the vehicle shall not slow down and should continue moving toward the intersection with the speed limit.

### Actual behavior

Even if the initially estimated ET is within a green phase (i.e., the vehicle can continue with speed limit and enter the intersection box within a green phase), the time buffer is added to it which causes the vehicle to slow down when it enters the TS area.

### Steps to reproduce the actual behavior

See the "Descriptive summary" section

### Related work

MishkaMN, 2022-04-15:
Potential fix includes on this branch
https://github.com/usdot-fhwa-stol/carma-platform/tree/fix/green-buffer

MishkaMN, 2022-04-26:
Switched from enhancement to anomaly as this behavior was not the intended design.

closed 2022-05-31 by msmcconnell