usdot-fhwa-stol/carma-platform #1737: Match MAP msg and vector_map.osm to be compatible

MishkaMN, 2022-04-26:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Current j2735 MAP.msg creator does not work with TFHRC vector_map.osm used in UC2, which is considerably old (from TIM use case). Center points created by the tool does not align with road center points in osm map.

There is a new .osm map from 03.10.22 (TFHRC_03.10.22.xodr.osm), which is in hd_map_cmc channel in slack that needs to be checked against the tool, and re-do the osm map if needed.

### Carma version where this issue was discovered
carma-system:4.0.0

### Expected behavior
Every lanes in MAP.msg must have corresponding lanelets in the HD map.

### Actual behavior
Some is not recognized.

### Steps to reproduce the actual behavior
Broadcast MAP.msg created by the tool from either of the intersection at TFRHC, and turn on logs for carma_wm_ctrl and start carma. 
Every lanes should correspond to correct entry, exit lanelet ids (verify with josm).

### Related work

msmcconnell, 2022-04-26:
The correct solution here would not be to change the HD map but instead identify the correct projection between the Open Streets Map used in the map tool and the projection used in the HD map. We can never expect a 1-1 correspondence.

msmcconnell, 2022-05-31:
Sufficiently resolved by above PRs