usdot-fhwa-stol/carma-platform #1743: carma-messenger DECA_M_PER_M incorrect value

afrye-leidos, 2022-04-28:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

In the j2735 - carma message convertors of carma-messenger, the distance fields are converted using the constant DECA_M_PER_M. This constant is defined as 10 in [units.hpp](https://github.com/usdot-fhwa-stol/carma-messenger/blob/develop/carma-messenger-core/j2735_convertor/include/j2735_convertor/units.hpp#L32) but there are actually 0.1 decameters in one meter. This issue is also present with DECA_MPS_PER_MPS and DECA_S_PER_S. 

The underlying math is correct, converting the j2735 fields in tenths to the carma fields in meters, but would fail if the constant is corrected [example](https://github.com/usdot-fhwa-stol/carma-messenger/blob/develop/carma-messenger-core/j2735_convertor/src/control_message_convertor.cpp#L114)

I see two solutions:
1. replace DECA_M_PER_M with DECI_M_PER_M, and leave the math unchanged because it is now correct
2. correct DECA_M_PER_M to be 0.1, and flip the math in each convertor (currently carma = j2735 / DECA_M_PER_M, would be fixed to be carma = j2735 * DECA_M_PER_M)

### Carma version where this issue was discovered

carma-messenger develop branch

### Expected behavior

### Actual behavior

### Steps to reproduce the actual behavior

1. Do this
1. Then do this...

### Related work

msmcconnell, 2022-04-28:
@afrye-leidos I believe those constants are misnamed and option 1. would be the correct solution. Feel free to apply the fix

afrye-leidos, 2022-04-28:
@msmcconnell sounds good, i'll make it happen

afrye-leidos, 2022-05-17:
This issue is solved, the above PR (carma-messenger 139) corrected the name of this constant.

closed 2022-05-17 by afrye-leidos