usdot-fhwa-stol/carma-platform #1778: Updated Radar Interface

GoodarzMehr, 2022-05-20:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

The "radar_msgs" package that was part of the [astuff_sensor_msgs](https://github.com/astuff/astuff_sensor_msgs) package is deprecated as of ROS Noetic. The replacement [radar_msgs](http://wiki.ros.org/radar_msgs) package uses new message formats, so CARMA Platform's radar interface (as well as the CARMA4 System Architecture document) should be updated accordingly.

### Carma version where this issue was discovered

develop after 4.0.3

### Expected behavior
N/A

### Actual behavior

N/A

### Steps to reproduce the actual behavior

N/A

### Related work

N/A

msmcconnell, 2022-05-23:
Thanks for catching this @GoodarzMehr 
I agree we should move to the updated interface. We will add an item to the backlog to update the docs/nodes.

msmcconnell, 2022-05-26:
Adding a comment for internal tracking. This is Jira ticket CAR-5443