usdot-fhwa-stol/carma-platform #1864: Platooning Plugin generate general lane change maneuver

saina-ramyar, 2022-07-29:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

The IHP platooning plugin generates a lane change if a solo vehicle needs to join a platoon in the adjacent lane. Currently, an assumption is that the target lanelet is an adjacent lane with left or right relation. This is not always the case, and the capability needs to be generalized to include other types of lane changes and relations.

### Carma version where this issue was discovered

### Expected behavior
The IHP generated lane change should not be limited to adjacent left or right lanelets.
### Actual behavior

The IHP generated lane change is currently limited to adjacent left or right lanelets.

### Steps to reproduce the actual behavior

1. Do this
1. Then do this...

### Related work

msmcconnell, 2022-07-29:
Above linked commit contains code which needs to be updated when resolving this issue

msmcconnell, 2022-09-20:
Resolved by above PRs

closed 2022-09-20 by msmcconnell