usdot-fhwa-stol/carma-platform #1939: Mobility Path Visualizer is crashing guidance container stack when receiving MP from another car

MishkaMN, 2022-09-20:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Recent fix due to UC1 made platform publish mobility path with updated timestamps even if the vehicle is stopping (which means coordinates x,y in MP are 0, 0). This interestingly crashes another vehicle's platform at the instant when that MP message is received by that 2nd vehicle. It was fine without that specific change before with two vehicles. We tracked down the source to mobility path visualizer, and with that node turned off, the vehicles operate without issue.  

Fix is introduced in this PR: https://github.com/usdot-fhwa-stol/carma-platform/pull/1936
Exact lines: 
https://github.com/usdot-fhwa-stol/carma-platform/blob/6ed83e7b1bbb8fd98d2774ca936a60cdda33be98/mobilitypath_publisher/src/mobilitypath_publisher.cpp#L90-L91

### Carma version where this issue was discovered

carma-system-4.2.0+ develop
### Expected behavior
See above.
### Actual behavior

### Steps to reproduce the actual behavior

Start platform on one platform with above fixes. 
Start another platform one a different vehicle with 4.2.0+ system and this vehicle's guidance stack will crash.

### Related work

MishkaMN, 2022-10-05:
fixed with above PR

MishkaMN, 2022-10-21:
Reopening this as there were more scenarios where the node could segfault, which is being addressed during UC3 integration testing.

msmcconnell, 2022-11-28:
Resolved by above PRs

closed 2022-11-28 by msmcconnell