usdot-fhwa-stol/carma-platform #1941: Turn off excessive logging from ray_ground_classifier

MishkaMN, 2022-09-20:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [X] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

On develop, autoware.auto publishes `std::cerr << "RayAggregator: Ray capacity overrun! Use smaller bins" << std::endl;` message constantly. This is related to package `ray_ground_classifier` here:

https://github.com/usdot-fhwa-stol/autoware.auto/blob/develop/AutowareAuto/src/perception/filters/ray_ground_classifier/src/ray_aggregator.cpp#L149

It is currently printing every 0.0001 second, which makes the log file huge. 
It either should be turned off, or it made toggleable by log level, or the frequency should be reduced. See below:

`1661204838.3450458 [lifecycle_component_wrapper_mt-3] RayAggregator: Ray capacity overrun! Use smaller bins
1661204838.3451259 [lifecycle_component_wrapper_mt-3] RayAggregator: Ray capacity overrun! Use smaller bins
1661204838.3452435 [lifecycle_component_wrapper_mt-3] RayAggregator: Ray capacity overrun! Use smaller bins
1661204838.3453438 [lifecycle_component_wrapper_mt-3] RayAggregator: Ray capacity overrun! Use smaller bins
1661204838.3454762 [lifecycle_component_wrapper_mt-3] RayAggregator: Ray capacity overrun! Use smaller bins
1661204838.3455884 [lifecycle_component_wrapper_mt-3] RayAggregator: Ray capacity overrun! Use smaller bins
1661204838.3456759 [lifecycle_component_wrapper_mt-3] RayAggregator: Ray capacity overrun! Use smaller bins
1661204838.3457479 [lifecycle_component_wrapper_mt-3] RayAggregator: Ray capacity overrun! Use smaller bins
1661204838.3458138 [lifecycle_component_wrapper_mt-3] RayAggregator: Ray capacity overrun! Use smaller bins
1661204838.3458776 [lifecycle_component_wrapper_mt-3] RayAggregator: Ray capacity overrun! Use smaller bins
1661204838.3459401 [lifecycle_component_wrapper_mt-3] RayAggregator: Ray capacity overrun! Use smaller bins
1661204838.3460112 [lifecycle_component_wrapper_mt-3] RayAggregator: Ray capacity overrun! Use smaller bins`

This actually maybe related to why carma has been slow overall especially with crystal computer hooked up for logging.

### Carma version where this issue was discovered

carma-system-4.2.0+ develop
### Expected behavior
see above.
### Actual behavior

see above.
### Steps to reproduce the actual behavior

### Related work

MishkaMN, 2022-09-20:
On the topic of excessive logging perhaps below logging should also be turned off or the level should be changed:

`platform_ros2                 | [carma_component_container_mt-9] 1663600640.432519820 | WARN | encode_mobility_operation_message:184 | Unacceptable recipient id value, changing to default
platform_ros2                 | [carma_component_container_mt-9] 1663600640.432606472 | WARN | encode_mobility_operation_message:220 | Unacceptable GUID, changing to default
platform_ros2                 | [carma_component_container_mt-4] 1663600640.447799680 | INFO | checkActiveGeofenceLogic:2039 | No active geofence llt ids are loaded to the WMBroadcaster
platform_ros2                 | [carma_component_container_mt-4] 1663600640.465225180 | INFO | checkActiveGeofenceLogic:2039 | No active geofence llt ids are loaded to the WMBroadcaster
platform_ros2                 | [carma_component_container_mt-11] 1663600640.479664540 | INFO | planTrajectory:407 | Guidance is not engaged. Plan delegator will not plan trajectory.`

msmcconnell, 2022-11-03:
Resolved by above PR