usdot-fhwa-stol/carma-platform #1988: Cannot engage CARMA vehicle when the cyclone_config parameters are added for ROS2 containers

dan-du-car, 2022-12-12:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

During the WFD tool testing, Telematic vehicle unit running in a separate PC needs to communicate with CARMA platform running on Spectra PC. To ensure the ROS2 nodes in those two PCs communicate with each other, the cycloneDDS config parameters are added to all the ROS2 containers: platform_ros2, ros1_bridge, other divers that use ROS2, telematic vehicle ROS2 Bridge. The Telematic ROS2 bridge was able to get all the available ROS2 topics from Platform and stream data from those topics. However, when login to the carma_web_ui to engage the vehicle, the UI engage button stays grey. Checked the /guidance/state topic, the state value equals to 2 which means system_ready. Normally the UI engage button turns green when the state value = 2. 

Remove the cyclone config parameters from the Platform ROS2 containers. The UI engage button turns green when the state = 2, and can engage the vehicle. However, without the cyclone config the Telematic ROS2 bridge cannot subscribe to the Platform ROS2 topics.

### Carma version where this issue was discovered

### Expected behavior
Telematic ROS2 bridge can subscribe to ROS2 topics and the vehicle can also engage
### Actual behavior

Telematic ROS2 bridge can subscribe to ROS2 topics but the vehicle cannot engage

### Steps to reproduce the actual behavior

NA

### Related work

afrye51, 2022-12-21:
Resolved. Some ros2 containers did not have /opt/carma/maps volume mapped, so they did not have the cyclone configuration file. When this volume is added, carma is able to engage successfully.

closed 2022-12-21 by afrye51

maefromm, 2023-01-26:
@afrye51 please attach the PRs associated to this change that you/anish did to fix this.