usdot-fhwa-stol/carma-platform #2006: LCI Tactical Plugin Trajectory Generation Improvement

MishkaMN, 2023-01-11:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [X] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

in UC3 (and UC2), on a tactical level, there is no decisions being made using lights. Only decisions are whether if
new TScase is same as before and last traj has more than 2 points and not within its last 1 sec trajectory, then use last trajectory
if not all above, use new trajectory.

Therefore if the plugin uses last trajectory for too long, the trajectory may have accumulated too much error as it is not updated for a while. So there needs to be check for error in distance. Since that's what is relevant for crossing the stop bar.
"Am I within 3m of my expected position at the current time"

Using last valid trajectory significantly reduces computing power, so it is preferred to still have it. See issue https://github.com/usdot-fhwa-stol/carma-platform/issues/1996
### Carma version where this issue was discovered

carma-platform 4.2.0+ (release/isetta)
### Expected behavior
lci tactical plugin follows most up to date and valid trajectory without using too much computing power. 
### Actual behavior

See above.
### Steps to reproduce the actual behavior

1. Do this
1. Then do this...

### Related work

maefromm, 2023-02-10:
@MishkaMN @JonSmet should this be closed now?

MishkaMN, 2024-01-03:
Resolved with above PR

closed 2024-01-03 by MishkaMN